arduino-libraries/ArduinoSound #6: WavPlayback example will not recognize any wav files

Protonerd, 2017-10-31:
I wired up the I2S MAX08357 breakout board to the MKZero and uploaded the WavPlayback from the https://github.com/arduino-libraries/ArduinoSound library. But somehow nothing useful happens.

If I use a wav file I get this:

Initializing SD card...initialization done.
wave file is invalid!

Disabling this check sheds some light on the issue:
Initializing SD card...initialization done.
Bits per sample = -1
Channels = -1
Sample rate = -1 Hz
Duration = 1 seconds
unable to play wave file using I2S!

This was with a mono, 8bit/sample wav file. OK, I tried a 16 bit mono, at least this file makes some sense to the parser...?
Bits per sample = 16
Channels = 1
Sample rate = 22050 Hz
Duration = 0 seconds
unable to play wave file using I2S!

If I go on and disable the sanity check this one file plays, not very clean, but at least something is coming out which is remotely like the sound I expect.

16bit stereo is again not recognized by the sketch...
Initializing SD card...initialization done.
Bits per sample = -1
Channels = -1
Sample rate = -1 Hz
Duration = 1 seconds
unable to play wave file using I2S!

To aggravate the problems, some 16bit/22050s/s/mono files will play, but not all of them, I cannot see the difference.

Any idea what do I do wrong?

sandeepmistry, 2017-11-02:
Hi @Protonerd,

> To aggravate the problems, some 16bit/22050s/s/mono files will play, but not all of them, I cannot see the difference.
>
> Any idea what do I do wrong?

At this time only stereo wave files are supported.

If you are still having issues, please attach the stereo wave files you're having issues with as a comment.

Protonerd, 2017-11-03:
Hi Sandeep, thanks for looking into this topic, I will attach example wav files as soon as I get home to my own PC. Hopefully this can be resolved. Do you have plans to give support to mono as well?

sandeepmistry, 2017-11-03:
> Do you have plans to give support to mono as well?

There is nothing scheduled to support mono files at this time. There will also be a performance hit, because the mono file needs to be converted to stereo at runtime to be played out the I2S interface.

Protonerd, 2017-11-03:
Hi Sandeep,

I attach one stereo wav file which refuses to play, I had to zip it with 7zip and rename the extension to zip otherwise GitHub will not accept it as attachment.
Thanks for looking into it, I really appreciate it!

[h_16b (1).zip](https://github.com/arduino-libraries/ArduinoSound/files/1442414/h_16b.1.zip)

Protonerd, 2017-11-06:
Hi Sandeep, any luck debugging the problem? I'm stuck with my project until this issue is solved... I appreciated your feedback. Maybe you can also send me a known-good wav file which plays.

sandeepmistry, 2017-11-16:
The wave file you provided has a `Subchunk1Size` value of 18, based on http://soundfile.sapp.org/doc/WaveFormat/ - 16 is expected for PCM.

I'll attach a known wave file shortly. Btw, the latest release of this library includes #7 which adds support for mono wave files.

sandeepmistry, 2017-11-16:
I've converted the wave file you provided with Audacity:
[h_16b-audacity.wav.zip](https://github.com/arduino-libraries/ArduinoSound/files/1479142/h_16b-audacity.wav.zip)

Protonerd, 2017-11-16:
Thanks Sandeep. In the mean time I also found wav files which are played correctly. So it's probably a header issue. I will try the new mono support, thanks a lot for the heads up. I will close it for now. Good jobs, thanks a lot for the support.
BTW, cam you point me to the place I can change the sample rate? I would like to change the pitch of the playback

closed 2017-11-16 by Protonerd