arduino-libraries/ArduinoSound #8: 24bit I2S mic support

xaviergh, 2018-01-31:
Hello, 
I've been using an I2S mic breakout board from Adafruit (SPH0645LM4H) https://www.adafruit.com/product/3421 and using the library successfully, however, while reading the datasheet (http://www.knowles.com/kor/content/download/5988/105771/version/4/file/SPH0645LM4H-BRevB.PDF) (page 3) I found that the actual output of the mic is 24bit rather than 32bit. Is there a possibility to add a 24bit option? 

I've been inspecting the library and I could gladly help adding the option so it's available for everyone.

Thank you.

Xavier

sandeepmistry, 2018-03-20:
Hi @xaviergh,

The part might work with the existing examples sketches that clock out 32-bits for the 24-bit ICS43432 I2S mic. Could you please try them with your mic?

24-bit is a bit tricky because DMA transfers only support 1, 2, or 4 byte samples.

JCWentzel, 2018-05-11:
It would be nice to get SDWaveFile.duration() back in milliseconds rather than seconds or as an option to get it back in milliseconds.