arduino-libraries/ArduinoSound #10: Flipping stereo channels

kapecNe, 2018-10-13:
I’m using Arduino MKR Zero with Adafruit UDA1334A breakout, when I play wav file which contain binaural audio, left and right channels are randomly switching their places. I’m using provided wav file example and reading wav file from sd card. Audio file is created in audacity and plays back on pc as intended. Any Ideas?

Rocketct, 2018-11-28:
Hi @kapecNecould, thanks for open the issue, could you please share a minimal sketch that reproduce the issue and the file that you are using?

sandeepmistry, 2019-01-30:
Closing this for now due to lack of feedback, please re-open with requested info if you are still interested.

aaitken203, 2021-12-06:
Hi,  hope you dont mind this old issue being re-opened.  i can create new if needed but I am experiencing the same issues described.

I have broken my setup into the most minimal to try to eliminate external vairables.  I am using an arduino MKR Zero and  2 mono MAX98357A i2s amplifiers but I am confident that they are setup correctly for them to real with left and right respectively.

I see the issue of channels switching even when using the example WavePlayback.ino sketch and the attached wav with the only change being to specify the new filename.  I have used the pins as defined in the example for connecting I2s devices.

I created that wav file to try to identify and debug the problem,  it is simply a stereo wav repeating a voice saying 'left' on the left channel and 'right' on the right.  [LR2.zip](https://github.com/arduino-libraries/ArduinoSound/files/7661560/LR2.zip)

During playback the channels will wander demonstrated by the 'left' voice appearing on the right channel etc.

I would appreciate any thoughts.  I will try to play with other sample rates etc to see if i can find any configurations in which this doesnt happen

cheers
a

aaitken203, 2021-12-10:
Thanks so much for re-opening.

I have been trying to delve into the problem with some success so far but I am hoping to get a logic analyser soon which should allow me to log the I2S to see whats physically happening on the lines.

A small breakthrough I have had whilst blindly stabbing in the dark is that by setting the I2S buffer to 5120 bytes at the start of startPlayback in AudioOutI2S.cpp (simply calling I2S.setBufferSize(5120) ) the channels no longer seem to switch _during_ playback however the channels will sometime alternate between playbacks.  Looks like there could be something buffer related possibly.  Please let me know if there is any tests I can conduct to help diagnose.

cheers
a

aaitken203, 2021-12-13:
I just recieved a logic8 so have managed to get some data which hopefully will be useful!  I have changed over to a UDA1334 stereo decoder to simplify the setup and for this test was playing a 2 channel wav file i made with a constant tone on one channel and silence on the other to highlight whats going on.

I have managed to get captures of the crossover showing both the analog audio output from the codec as well as the digital I2S data simultaneously.  There are some CSV and Salaea files in the attachement.

I have attached a zip with a bunch of screen grabs and a pdf to try to explain the observations.
[I2SChannelSwap.pdf](https://github.com/arduino-libraries/ArduinoSound/files/7704713/I2SChannelSwap.pdf)
[I2sChannelSwap.zip](https://github.com/arduino-libraries/ArduinoSound/files/7704719/I2sChannelSwap.zip)

Is there anything else i could test to gather more info?  perhaps 44.1Khz stereo is pushing the M0 a little too hard??

cheers
a

aaitken203, 2021-12-23:
I have been playing about putting GPIO toggles on various points such as swapping of the double buffers but cannot seem to find a correlation.  I have also tried exporting WAVs from other applications and even different OS's to try to rule out peculiarities in the files but I cant get anything higher than 16kHz audio stereo audio to play without the channel swap.

The page here (https://www.arduino.cc/en/Tutorial/ArduinoSoundWavePlayback) does specify 44.1khz signed 16bit stereo so as far as I can tell I'm working within the intended scope of the library.

Is any one able to confirm that they have managed to successfully get 44.1KHz (or even 32KHz) 16bit stereo wavs playing on a SAMD21 with this library without the channels flipping?  If so could they provide the wav file and info on the setup they used so I can work back from a working example? 

cheers
a