arduino-libraries/ArduinoSound #14: Support nRF52 based boards and I2S amp

Protonerd, 2019-04-12:
Hi Sandeep,
I've seen you released quite many great additions to support boards with nRF52 from Nordic, like the Adafruit BLE Feather.

Do you plan to adapt this library to support the nRF52 platform too?

The chipset seems to be very popular due to its powerful architecture and flexible pin configurations, not to mention the EasyDMA also for I2S. I already use the BLE Feather with a MAX98357A Breakout and this library offers the infrastructure to read data from SD-card and stream it to an audio amp. Frankly I do not know how much effort it would be to tweak the code to support nRF52, but since you already have extensive experience with that platform and knowing your skills from other libraries for nRF52 you are the best guy for the job!

facchinm, 2019-04-12:
Hi @Protonerd ,
this library is currently maintained by Arduino (even if Sandeep is the main contributor :slightly_smiling_face: ). 

We usually do not write code for unofficial Arduino boards because we have no control over them, but pull requests are always welcome :wink:

closed 2019-04-12 by facchinm

Protonerd, 2019-05-23:
I hope this topic will be worth a revisit in light of the new line of Nano boards using the nRF52 chips :)
[https://blog.arduino.cc/2019/05/17/whats-new-at-maker-faire-bay-area-2019/]

jbkuma, 2020-02-04:
This topic is still relevant, hopefully it gets some attention, or at least a new library can be recommended.