arduino-libraries/ArduinoSound #16: Change amplitude example to output values in dB

sandeepmistry, 2019-04-17:
Suggestion from @tigoe.

ashutoshshrm529, 2020-03-17:
@sandeepmistry could you please elaborate on this? What unit does the output show currently?

MisterAwesome23, 2020-03-21:
I would like to fix this issue. Is anyone on it already?

per1234, 2020-03-21:
@MisterAwesome23 there is an existing pull request to fix this:
https://github.com/arduino-libraries/ArduinoSound/pull/24
Feel free to test it out and review it.

MisterAwesome23, 2020-03-22:
I feel it can be improved minutely. 

- No need to use abs(), two reasons, technically sound can be negative decibels also, the ```amplitudeAnalyzer.read()``` should ideally return a non negative value so no need of abs()

Should I create my own pull request or suggest the changes there?

per1234, 2020-03-22:
@MisterAwesome23 I think it would be best to start by suggesting your changes on the existing pull request. The reason is that it is more convenient for the repository maintainers to have only a single pull request to review and merge, rather than having to evaluate two different PRs to decide which should be merged and which closed.

If there is a situation such as you and the PR author having differences of opinion, then it would be great if you would submit a PR.

Thanks!

MisterAwesome23, 2020-03-22:
Sure. Makes sense to keep a single pull request. I'll suggest my changes there. Thanks :)