arduino-libraries/ArduinoSound #28: Cue function not working

Iqwertz, 2020-12-25:
Don't know if I understood the cue() feature correct, but shouldnt it play a Song from the given time on? When I call it, it only replays the song from beginning!

Thats a part of my code: 

  ```
void audioCue(int t){
    if(t>0 && t<currentAudioFile.duration()){
      currentAudioFile.cue(t);
      Serial.println("Set playback to:");
      Serial.println(currentAudioFile.currentTime());
    }else{
      Serial.println("Cued Time not valid");
    }
}
```
currentAudioFile is a global var holding the currently played wav file (type: SDWaveFile)

Iqwertz, 2020-12-26:
I looked into the code and it seems that there are a view errors! I listed them in this Forum Post: https://forum.arduino.cc/index.php?topic=720007.msg4837911#msg4837911

Iqwertz, 2020-12-26:
Okay because I am impatient! I fixed the code and created a Pull request. Please review the request and if accepted I will close this issue!

aentinger, 2022-06-30:
Fixed by #29 -

closed 2022-06-30 by aentinger