arduino-libraries/ArduinoSound #31: Recording .wav files using I2S MEMS microphone SPH0645 and Arduino MKR1010

Ajiket, 2021-02-12:
Respected All,
I want to record 10-20 seconds audio files in .wav format. The audio source is coming from the earpieces of a Stethoscope to convert a conventional stethoscope into a digital stethoscope.
I have connected the SPH0645LM4H-B Adafruit I2S digital microphone with Arduino MKR1010.

With the help of https://github.com/arduino-libraries/ArduinoSound/issues/2#issuecomment-271968591 by @sandeepmistry 
I am able to see visually the audio data in serial plotter. 

I want to store this audio recordings in .wav format in (say my computer drive) for further signal processing in Matlab.

Thanking You,
Ajiket

per1234, 2021-02-12:
Hi @Ajiket. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.

closed 2021-02-12 by per1234

Ajiket, 2021-02-12:
> Hi @Ajiket. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over [...]

Thank You @per1234 , i posted my issue in the Audio Sound processing and generation Arduino forum.